mate-desktop/mate-sensors-applet #22: expose the ability to hide the temperature units of the sensors

sunweaver, 2015-03-09:
A feature request from a Debian use (DebianBug#778756):

"""
Could you please expose the ability to hide the temperature units of the
sensors in the applet? This would help a lot for those that have limited
space on their screens.
"""

sc0w, 2016-07-12:
https://github.com/mate-desktop/mate-sensors-applet/blob/master/sensors-applet/active-sensor.c#L589

I can't find the gsettings key, where is it?

If I set the var hide_units to true, it works

I think it is a good idea to put a checkbox on preferences

raveit65, 2016-07-12:
I don't think there is a gsettings key for it, btw. i didn't find them.

sc0w, 2016-07-13:
I saw this:

http://wiki.mate-desktop.org/docs:mate-sensors-applet

> Hide Units
> 
> The units for sensor values can also be hidden if desired by setting the boolean value /apps/sensors-applet/hide-units within the user's MateConf database. This can be achieved by the following command, where the value is specified as either true or false and is specified as the last parameter. In this case is the units will be hidden (the value is set to true):
> 
> mateconftool-2 -s /apps/sensors-applet/hide-units -t bool true
> 
> These values can also be set using the Configuration Editor.

mate-sensors-applet was migrated to gsettings with this commit: https://github.com/mate-desktop/mate-sensors-applet/commit/fbb7e89c4eab794cc9151e119c7ee94c9494dd7d

info-cppsp, 2017-09-04:
more info:
in CLI:
gsettings list-relocatable-schemas
found org.mate.sensors-applet.sensor

http://wiki.mate-desktop.org/docs:gsettings

use d-conf editor to find the right path (object)
org/mate/panel/objects/object-2

[image]

get current setting
gsettings get org.mate.sensors-applet:/org/mate/panel/objects/object_2/ hide-units
(should return false)

set hide-units
gsettings set org.mate.sensors-applet:/org/mate/panel/objects/object_2/ hide-units true

restart -> not working

raveit65, 2017-10-25:
Ok, something like this works
```
gsettings set org.mate.sensors-applet:/org/mate/panel/objects/object-26/prefs/ hide-units true
```
But the hide-unit key exists only in dconf-editor after i set it via cli.
So, this isn't very user friendly and i'd like to see the setting in preferences.
Also the key is marked with 'No scheme found` in dconf-editor for some reason.
Looks like the implementation of the key is a bit worse.

info-cppsp, 2017-10-25:
yeaaah...
see #52 
That is probably bc the schema is a relocatable one.
??

raveit65, 2017-10-25:
All this keys you will see only in dconf-editor if you set them in preferences or via cli.
```
[rave@mother ~]$ cat /usr/share/glib-2.0/schemas/org.mate.sensors-applet.gschema.xml 
<schemalist gettext-domain="mate-sensors-applet">
  <schema id="org.mate.sensors-applet">
    <key name="font-size" type="i">
      <default>0</default>
      <summary>Font size in points</summary>
    </key>
    <key name="hide-units" type="b">
      <default>false</default>
      <summary>Specify if show or hide sensor units</summary>
    </key>
    <key name="display-mode" type="i">
      <default>1</default>
      <summary>Set if display icons or text labels</summary>
    </key>
    <key name="layout-mode" type="i">
      <default>0</default>
      <summary>Set value beside or below label</summary>
    </key>
    <key name="temperature-scale" type="i">
      <default>1</default>
      <summary>Show temperature in Kelvin, Celsius or Fahrenheit </summary>
    </key>
    <key name="display-notifications" type="b">
      <default>true</default>
      <summary>Whether to display notifications</summary>
    </key>
    <key name="timeout-delay" type="i">
      <default>2000</default>
      <summary>Delay (in ms) between refreshes</summary>
    </key>
    <key name="graph-size" type="i">
      <default>42</default>
      <summary>The size of the graph in pixels (width if horizontal, height if vertical)</summary>
    </key>
    <key name="sensors-list" type="as">
      <default>[]</default>
      <summary>List of sensors</summary>
    </key>
  </schema>
</schemalist>
```
Not sure it's because a relocatable one.
Ie. clock-applet ships also some keys in dconf-editor prefs folder after installation.
Maybe @monsta can enlighten us about gsettings/glib stuff in m-s-a?

info-cppsp, 2017-10-25:
No I meant, 'No schema' shows up in d-conf, bc the schema is relocatable.
[image]
That is what you mean, right?
(this is the clock-applet)

But you made a good point!
clock-applet prefs are visible
That just means, that I (or sy else) have to check, why the options don't show up.
Maybe we won't need the PR after all.

raveit65, 2017-10-25:
Offtopic on
```
commit fc89b967f91eec5db4ddfe2187b60119e3e9bd28 (origin/master, origin/HEAD, master)
Author: Unknown <info@cppsp.de>
Date:   Thu Oct 19 11:00:04 2017 +0200

    remove code duplication 01
```
Unknown is a really nice name in git history :-)
You should check your .gitconfig in home folder.

Offtopic off

info-cppsp, 2017-10-25:
https://github.com/GNOME/dconf-editor/blob/master/editor/registry-info.ui
search for 'No schema available'
it says
"No schema available. Dconf Editor can’t find a schema associated with this key. The application that installed this key may have been removed, may have stop the use of this key, or **may use a relocatable schema** for defining its keys."

info-cppsp, 2017-10-25:
@above
I was wondering about that myself...

info-cppsp, 2017-10-25:
OFF
can you please fix the history for me?
just use 'info-cppsp'
Thanks.
I have started using gitkraken, but something is wrong with it and now I am Unknown...

raveit65, 2017-10-25:
I will change Author in existing open PRs before merging.
The commits in master i can't change anymore, sorry.
My .gitconfig looks like this
```
[user]
	name = raveit65
	email = mate@raveit.de
[help]
	autocorrect = 1
[push]
	default = simple
[alias]
  pu = !"git fetch origin -v; git fetch upstream -v; git merge upstream/master"
```
or
https://git-scm.com/book/en/v2/Getting-Started-First-Time-Git-Setup

info-cppsp, 2017-10-25:
does this help?
https://kevin.deldycke.com/2010/05/how-to-fix-bad-commit-authorship-git/

all the info is correct, even in the program.
I have contacted support.

raveit65, 2017-10-25:
I use command line
```
example:

git rebase -i <commit-id>

choose edit an existing commit

git commit --amend --author="name <email address>" --no-edit

git rebase --continue
```

Anyway you can edit .gitconfig in your home folder by hand.

raveit65, 2017-10-25:
fixed https://github.com/mate-desktop/mate-sensors-applet/commit/9912a19d75bce7b44dc5f5c259bafa312b482afc

closed 2017-10-25 by raveit65

info-cppsp, 2017-10-26:
OFF
I think I got the name problem sorted.
If you would please remove the off topic comment.

info-cppsp, 2017-10-26:
about being able to edit configuration not set before
https://bugzilla.gnome.org/show_bug.cgi?id=755925
It is a relocatable schema problem after all. It may be fixed one day...

info-cppsp, 2017-10-26:
This lists all keys:
gsettings list-recursively org.mate.sensors-applet:/org/mate/panel/objects/object-16/prefs/