mate-desktop/mate-sensors-applet #35: Hdd temperature in mate-sensors-applet is imprecise

acrobat1, 2015-08-17:
In  mate-sensors-applet, starting from MATE 1.6 up to MATE 1.10 , the hdd temperature on my computer is never correct: regardless of which corrections are applied in *preferences-->sensors-->hddtemp*, the value displayed in the applet will be again largely off at every reboot.
Temperatures were reported correctly  in older MATE versions and in Gnome.

acrobat1, 2015-10-09:
As I wrote here https://github.com/mate-desktop/mate-sensors-applet/issues/11# after installing MATE 1.8 in  Debian 7,  for the first time I'm seeing see the applet reporting correct temperatures for hard disks.

dcagigas, 2016-07-27:
I have installed Linux-Mint 18 that uses 1.14.1 version. Using Linux-Mint 17.3 no problem to use udisk and show hard disks temperatures. However, now Mint 18 uses udisk2 and that seems to be the problem. I'm not able any more to see/select hard disks temperatures. I can check hard disks temperatures using hddtemp and the command line. 
But it would be nice to check it using the mate-sensors-applet like in Linux-Mint 17.3.
Is this a bug?

raveit65, 2016-07-27:
Well, if Linuxmint dropped udisk than i suggest to file out a report against linuxmint.
We can't help you.
Btw. in fedora rawhide (next release) udisk2 is replaced by something else, which comes from freedesktop.
So porting to udisk2 does make any sense, as fedora changes will hit your distros with a delay of 1-2 years ;)

monsta, 2016-09-19:
@dcagigas: hddtemp isn't running as daemon by default, at least in Debian/Ubuntu/Mint. Run ```sudo dpkg-reconfigure hddtemp``` and set it to run as daemon. Then you'll be able to select it in the applet preferences.

monsta, 2016-09-19:
Since the original bug is already solved, it's time to close the report.

closed 2016-09-19 by monsta

dcagigas, 2016-09-19:
Thanks. Yes now it works. Finally, I can see the hard disks temperatures in the applet.