mate-desktop/mate-sensors-applet #36: Udisks instead hddtemp ?

kajzersoze, 2016-03-23:
Any chance of using udisks instead of hddtemp ?
I get temperature just fine in terminal if I type :
"udisks --show-info /dev/sda | grep temperature-celsius-2 | cut -c 52-53"

No need to mess with hddtemp IMO
At least option for adding custom sensor (by entering bash command) would be great.

monsta, 2016-03-24:
I'm not sure... but isn't udisks deprecated these days in favor of udisks2?

raveit65, 2016-03-24:
yes, udisk is deprecated and distros like rhel7/centos7 ship only udisk2.
There are some distros, ie. fedora who ship both versions.
But /usr/bin/udisks is from udisks-1.0.5 here.
```
[rave@mother ~]$ sudo yum-deprecated provides /usr/bin/udisks
Yum command has been deprecated, use dnf instead.
See 'man dnf' and 'man yum2dnf' for more information.

Loaded plugins: auto-update-debuginfo, copr, remove-with-leaves, show-leaves,
              : versionlock
udisks-1.0.5-4.fc24.x86_64 : Storage Management Service
Repo        : fedora
Matched from:
Filename    : /usr/bin/udisks
```
So, using udisks is not an option.

raveit65, 2016-03-24:
> At least option for adding custom sensor (by entering bash command) would be great.

You can use the command-applet for this, which displays the result of a bash command in the panel.
In the past i used it to read disk sensors from a hardware-raid-controller.

kajzersoze, 2016-03-24:
Okay, I understand, I guess then that with udisks2 it's not that easy to read temperatures.

As for command-applet... seems like it's having some problems with cut and grep, more likely with pipe.
Without using pipe it works fine, but with pipe it just displays #
Maybe I'm doing something wrong, any tips would be welcome.

kajzersoze, 2016-03-24:
Solved, I wrote a bash script with what I need and then just start that script with command-applet.

Thanks for all the help !

closed 2016-03-24 by kajzersoze