mate-desktop/mate-sensors-applet #41: mate sensors applet dont respect its position.

javashin, 2017-03-15:
hello the mate-sensors-applet-1.18.0 is not showing in the correct place in the mate panel i installed just with configure --prefix=/usr and i had to downgrade to the 1.16.0 version in order to make it work like it has to be here is a screenshot in the correct place with old verision inside full mate 1.18
[image]

javashin, 2017-03-15:
and here is the mate-sensors-applet 1.18.0 in a way other level with the same configuration as the other screenshot 
[image]

javashin, 2017-03-15:
this is gentoo with mate 1.18 gtk3 all compiled with:
CFLAGS="-march=westmere -mtune=westmere -O2 -pipe -fstack-protector-strong"
everything is working like it has to be only mate-sensors applet have this issue so far . thanks in advance.

flexiondotorg, 2017-03-15:
I don't know what you're trying to describe here? I can't see any obvious problem when comparing the two screenshots.

javashin, 2017-03-16:
cool

monsta, 2017-03-17:
Confirmed. Looks like a regression from 0392b7a56d88153f5305bf2c50c8cb958572ee9f. :disappointed:

raveit65, 2017-03-18:
We need to set valign=GTK_ALIGN_CENTER to the whole GtkGrid, than every sensor is in the middle of the panel. The default GTK_ALIGN_FILL isn't useful here.
You can test this in runtime with gtk3-inspector and set up this in inspectors properties page for the grid.
I guess for a vertical panel we need halign=GTK_ALIGN_CENTER.

lukefromdc, 2017-03-18:
confirmed

monsta, 2017-03-19:
@javashin: fixed in https://github.com/mate-desktop/mate-sensors-applet/commit/6dd3a48122004ac8bec2264ac513a6e2b7b760a1, you can make a patch from that commit (we will release 1.18.1 later).

closed 2017-03-19 by monsta