mate-desktop/mate-sensors-applet #43: fan rpm inverted?

herbie63, 2017-05-02:
it looks as if in version 1.16.0 on Mint 18.1-64 the fan rpm is calculated or displayd in a wrong way:
Under CPU stress the fan rpm audibly increases but the displayed rpm value decreases.

info-cppsp, 2017-09-04:
on mint 18.2 64 with msa 1.19 the i2c-sys plugin fan1 sensor rpm increases from 818 to 1119 or so when using prime 95.
Stopping the fan with my hand doesn't seem to do much...
Which plugin do you use?

herbie63, 2017-09-04:
as mentioned above: MATE Sensor Applet 1.16.0

info-cppsp, 2017-09-05:
not which applet version, but which sensor plugin.
[image]
I am using udisks2 and i2c-sys. You?
Also, is it an option for you to update the plugin to 1.18 or build it from the source here?
Or at least try a live mint 18.2 to see if it is working there?

herbie63, 2017-09-05:
OK :-) The machine in question (an old Lenovo x121) has the libsensors and ibm-acpi but only read out the libs fan1 = CPU fan, temp1 = CPU temp and  temp7 = Mainboard temp. CPU fan and CPU temp are also available vie the ibm-acpi wirh identical results.
I only use that netbook from time to time. But yes an update surely is a good idea - I'll give it a try the other days.
Thanks!
Herbert

info-cppsp, 2017-12-03:
have you tried the update yet?

I have found something:

[image]

This may be a bit cumbersome, but if you get into dconf-editor, under the path /org/mate/panel/objects/ under one of the object-nr/prefs you see these ugly hashes. These are the sensors used by msa. On the picture you can see my CPU temp sensor's data. Under path there is `/sys/devices/platform/it87.656/hwmon/hwmon1/temp2_input`.
That is path on my system. So if I say:
`cat /sys/devices/platform/it87.656/hwmon/hwmon1/temp2_input`
I get the temperature reading. (In my case 31000 for 31C.)

You could try this with your fan sensor to see if it is read correctly at all.

info-cppsp, 2018-01-26:
I have reviewed the imb-acpi plugin.
```
cat /proc/acpi/ibm/fan
```
I get a fan speed, that looks correct, on an old laptop with msa v 1.12 on an up-to-date mint 17.3.

@raveit65 
If there is no reply, please close this in a few days.

raveit65, 2018-01-26:
@herbie63 
Does the issue still exists?

herbie63, 2018-01-27:
yes an no. It still exists but I don't care anymore - think about switching to Win with that device.
So I think this thread can be closed.

closed 2018-01-27 by herbie63