mate-desktop/mate-sensors-applet #45: Graphs Invisible (may be present but greyscale)

glimmung, 2017-06-05:
On a current Ubuntu Mate version, selecting a graph for a given sensor is ineffective.

Peering very closely at the screen, it seems that the graph may be there but in shades of grey so close as to be indistinguishable
[image]
 (tested on multiple machines to sanity-check)

BenLand100, 2017-06-15:
I'm seeing the same thing on version 1.18.2. Initially I thought this might be theme related but that seems not to be the case.

info-cppsp, 2017-09-04:
changing the bg color from black to white reveals that the graphs are drawn in black
active-sensors.c l279
cairo_set_source_rgb(cr, 0.0, 0.0, 0.0); -> cairo_set_source_rgb(cr, 1.0, 1.0, 1.0);

info-cppsp, 2017-09-04:
in active-sensors.c
lines 314,315,316, 320,321,322 have a division (/ 65535.0).
this is not needed.
[color.txt](https://github.com/mate-desktop/mate-sensors-applet/files/1274093/color.txt)

here is a diff -u patch

info-cppsp, 2017-09-11:
I have tried to follow this:
http://wiki.mate-desktop.org/contribute
I hope I did everything right.
Please review and merge.

monsta, 2017-09-12:
Ok, fixed by https://github.com/mate-desktop/mate-sensors-applet/pull/49.

closed 2017-09-12 by monsta

monsta, 2017-10-25:
The fix is now released in version 1.18.3.