mate-desktop/mate-sensors-applet #46: [GTK+3] 1.18.1 not save order of sensors

spvkgn, 2017-06-08:
MATE Sensors-applet 1.18.1 does not save order of sensors from two sources - libsensors and nvidia in my case. When I restart session or `mate-panel --replace` the order resets to default state.

monsta, 2017-06-19:
Do you mean it somehow worked in previous versions? :open_mouth:

spvkgn, 2017-06-19:
I used 1.8.2 before 1.18 (on same hardware), did not see this issue there.

flexiondotorg, 2017-06-19:
I don't think it worked since the migration to gsettings.

monsta, 2017-06-19:
Yes, and it's reported in https://github.com/mate-desktop/mate-sensors-applet/issues/11 which goes way back to 1.6 (the first MATE release with GSettings).
That's why I asked. Can you test it with 1.8.x again to make sure?

info-cppsp, 2017-09-04:
I confirm that it doesn't work with 1.19 on mint 18.2 64

info-cppsp, 2017-09-04:
As far as I can see, there is no code to actually save sensors in msa.
I have found the following file in gsa:
https://sourceforge.net/p/sensors-applet/code/ci/master/tree/sensors-applet/sensors-applet-conf.c
This may do the trick.

How should someone store an ordered list of sensors in gsettings?
prefs_dialog_sensor_down_button_clicked() calls sensors_applet_reorder_sensors() which calls sensors_applet_pack_display() in sensors-applet.c line 451
I guess it would be ideal to save the sort in there.

info-cppsp, 2017-09-10:
I have tried the ubuntu 17.04 64bit iso with the gnome-fallback option, which has the gnome sensors applet. (as a live cd) logging out and back in the sensor order stays as set and indeed an array is visible from dconf editor.
so basically this function just has to be ported from gsa.

info-cppsp, 2017-09-11:
sensors are saved in sensors-applet-settings.c sensors_applet_settings_save_sensors()
called from prefs-dialog.c prefs_dialog_close()

sensors are loaded with:
sensors-applet.c sensors_applet_init() calls sensors-applet-plugins.c sensors_applet_plugins_load_all() which calls sensors_applet_plugin_init() on all plugins that return the sensors
on each sensor sensors-applet.c sensors_applet_add_sensor() is run

so far it is not clear to me whether or not the sorting is saved, but not loaded OR not saved at all

info-cppsp, 2017-10-23:
Note that this problem has been with us since:
https://github.com/mate-desktop/mate-sensors-applet/commit/fbb7e89c4eab794cc9151e119c7ee94c9494dd7d

raveit65, 2017-11-05:
fixed with https://github.com/mate-desktop/mate-sensors-applet/commit/c5bce79ac7c654716cd8e1d69882a67053d469ef

closed 2017-11-05 by raveit65