mate-desktop/mate-sensors-applet #47: udisks plugin needs to be ported to udisks2

pacho2, 2017-07-24:
It seems udisks plugin is still relying on dead and obsolete udisks1... it should be ported to udisks2 if possible as udisks1 is being removed from major distributions:
https://github.com/mate-desktop/mate-sensors-applet/blob/master/plugins/udisks/udisks-plugin.c

Thanks

monsta, 2017-07-24:
Yes, it needs to be done.

info-cppsp, 2017-08-29:
After 3 days of pain, I have a plugin for udisks2.
It can display the temp once, after adding it to the mate-panel.
I guess I shall upload it after it can refresh as well.

info-cppsp, 2017-08-30:
I have made a new repo, bc I didn't want to spam this one.
I am new to git as I am mostly using mercurial.
https://github.com/info-cppsp/mate-sensors-applet
I seems to work, but I would like to ask anyone reading this to read, check, test the code and give me feedback.

raveit65, 2017-08-30:
Tip, you can use 'git rebase -i *last-commit-before-you-want-to change-something*' for editing/squashing/droping existing commits. (i= interactive)
I did that local and i have only i commit as a patch for testing. (I ignore README changes for the moment).
```
[rave@mother mate-sensors-applet]$ git format-patch  68356eef7b581f680604e8fd17f8aca23e1d5751
0001-added-udisks2-plugin.patch
0002-updated-readme-with-a-couple-of-infos.patch
```
This will make it easier for a possible pull reguest later.

raveit65, 2017-08-30:
I can't load the applet any more to panel!
Build with 
configure \
    --disable-static \
    --disable-schemas-compile \
    --enable-libnotify \
    --with-nvidia \
    --disable-udisks
for build with udisk2 only.
```
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007f971c09645d in udisks2_plugin_get_sensors (sensors=0x7ffd2772ddf0) at udisks2-plugin.c:445
445					g_debug ("Cannot obtain data for device: %s\n"
[Current thread is 1 (Thread 0x7f9738cb3a80 (LWP 22117))]
```
complete stacktrace at
https://www.dropbox.com/s/ca2t1sv0wz2bgfq/backtrace-m-s-a?dl=0

raveit65, 2017-08-30:
Btw. device is a SSD from OCZ. Other disks are behind a 2 hardware-raid-controller, so they won't read at all.

info-cppsp, 2017-08-30:
thanks for the tip. I gotta leave, so I won't be of use until tomorrow. sry.
I forgot to write that you can use a program called d-feet to access DBus with a GUI.
[image]
this is the smart interface that you would want to see.
I shall get back to you later.

info-cppsp, 2017-08-30:
you could turn on the debug in the plugin and see where it gets stuck
just remove // from the define debug line and look at syslog
/var/log/syslog

raveit65, 2017-08-30:
oh, it looks like it fails with the little config device from my main Hardware-raid-conroller (with / partitions).
```
Thread 1 (Thread 0x7f9738cb3a80 (LWP 22117)):
#0  0x00007f971c09645d in udisks2_plugin_get_sensors (sensors=0x7ffd2772ddf0) at udisks2-plugin.c:445
        id = 0x7f9710052160 "Config---Disk-0-CP-5723_Port_0_1_0_J"
        model = 0x7f97100517a0 "Config   Disk 0"
        smartenabled = 0
        temp = 0
        propdict = 0x7f9710054920
        propdict2 = 0x7f9710051240
        info = <optimized out>
        iter = {x = {140286785830288, 26, 12, 0, 491560395275, 491558639920, 0, 3579507750, 491556253184, 140287466129061, 140287470493696, 0, 140286987165824, 140287466127586, 491557263824, 491537722904}}
        key = 0x72733c8c10 "/org/freedesktop/UDisks2/drives/Config___Disk_0_CP_5723_Port_0_1_0_J"
        value = 0x7f9710054980
        proxy = 0x7273527f80 [GDBusProxy]
        error = 0x0
        managed_objects = 0x7f9710093d90
        sensors = 0x0
```
[image]
On this drive are the raid array informations stored,
I can't disable that.

Good is that we run in a very special case issue.
Such a device should simply be ignored.

raveit65, 2017-08-30:
Hmm, on my notebook with only one SSD it tries to read the cd- drive and crashes with the same error.
```
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007f032184545d in udisks2_plugin_get_sensors (sensors=0x7ffca83f88a0) at udisks2-plugin.c:445
445					g_debug ("Cannot obtain data for device: %s\n"
[Current thread is 1 (Thread 0x7f0335a1f280 (LWP 1415))]

Thread 1 (Thread 0x7f0335a1f280 (LWP 1415)):
#0  0x00007f032184545d in udisks2_plugin_get_sensors (sensors=0x7ffca83f88a0) at udisks2-plugin.c:445
        id = 0x7f0318042900 "HL-DT-STDVDRAM-GT32N-KZHAA5B0702"
        model = 0x7f0318042720 "HL-DT-STDVDRAM GT32N"
        smartenabled = 0
        temp = 0
        propdict = 0x7f0318045a90
        propdict2 = 0x7f0318041e30
        info = <optimized out>
        iter = {x = {139651264595344, 13, 11, 0, 522451267715, 522449059136, 0, 3579507750, 522447216704, 139651757905573, 139651762270208, 0, 139651423948928, 139651757904098, 522450891456, 522443279896}}
        key = 0x79a45eb600 "/org/freedesktop/UDisks2/drives/HL_DT_STDVDRAM_GT32N_KZHAA5B0702"
        value = 0x7f0318045af0
        proxy = 0x79a4551680 [GDBusProxy]
        error = 0x0
        managed_objects = 0x7f031804c590
        sensors = 0x0
```
stacktrace
https://www.dropbox.com/s/yn183l8vxfynwe4/backtrace-m-s-a-notebook?dl=0

Same as before, a cd drive should simply ignored.
...this is a bit too much alpha for me...

info-cppsp, 2017-08-31:
I gather the following problems from your comments:
1. segfault on raid controller
That is bad, it should just be ignored as you say. However that could be another feature request. Can you access smart data in that Config__Disk_0... structure?
2. I don't have a cd drive, so that's your problem. LOL :P
ok, i have a dvd-drive that I can attach to my computer. It seems from what you wrote that segfault occurs, when a drive doesn't have a temperature.

The plugin has two main parts:
1. gather drives from DBus/UDisks2 and get their temp (initial value)
2. read temp from drives found at every refresh

Segfault comes in the 1st (obviously).
I haven't tested the plugin with devices that don't have temp before, as I have none attached, but I shall correct that now.

info-cppsp, 2017-08-31:
ok, I have committed some changes to the udisks2-plugin.c file
I am going to read more about rebase later, bc I don't understand it fully yet.
The problem was just that when I was trying to get the temp but couldn't, I tried to write out an error message that didn't exist. I copied the wrong error handling code...
If you would try again. Thank you.

info-cppsp, 2017-08-31:
ok, now I have cleaned up the code more
consider it a beta :)

raveit65, 2017-08-31:
> segfault on raid controller
That is bad, it should just be ignored as you say. However that could be another feature request. Can you access smart data in that Config__Disk_0... structure?

Well, this isn't a real drive, only a 8GB partition where are the Data of the Raid Array is stored.
I will never touch this partition as if something goes wrong the whole raid structure of the raid is damaged.

raveit65, 2017-08-31:
Ok, the applet doesn't crash anymore on both boxes.
But my SSDs have only a dummy temp sensor which shows always 30 C  ;-)
So i can't really say if the temp would be correctly.

info-cppsp, 2017-09-01:
OK.

Great!
Well, then this plugin doesn't help you much...

info-cppsp, 2017-10-19:
I guess you can close this as well, as it is solved by PR #50 
Thanks.
I have tried to document the plugin as much as possible, but if someone should have a question in the future, (s)he can ask it here or send me an email.

sc0w, 2017-10-19:
@info-cppsp if you like, in future PRs, you can put in the last line of the commit message: 

**Fixes #numberissue**

and the issue will be automatically closed after the merge

info-cppsp, 2017-10-19:
TIL
Thanks!

monsta, 2017-10-24:
Well, it works, but it seems like it doesn't update right away when temperature changes. Several minutes pass before it updates. I manually checked the temperature with hddtemp to compare.

Anyway, I'll close this report as the feature request is filled.

closed 2017-10-24 by monsta

info-cppsp, 2017-10-24:
https://github.com/info-cppsp/mate-sensors-applet

"I would like to note one major point:
The udisks plugin and the new udisks2 plugin asks DBus for the HDD temperature every refresh time seconds.
The default is 2s. However the smart data is updated every 10 mins.
There is a function to force this data update, but I don't think it is necessary.

I would also note that polling DBus every 2 seconds sound like a waste of computing power. ;-)

Before, the non solution for this non problem was to use a 'changed' signal from DBus as well to update the temp.
I better say that as of now it is unclear to me why or whether or not it is a priority to display a temp change 
right after it has happened and not within max. 10 seconds.
Therefore I haven't rewritten that part of the code."

info-cppsp, 2017-10-25:
these pics are from mint 17.3 32 bit
what you see here are the signals from /org/freedesktop/UDisks/devices/sda
[image]

what you see here are the signals from /org/freedesktop/UDisks
[image]

In udisks-plugin.c line 239 the signal "Changed" is used.
I couldn't find that signal in UDisks2.
In fact, there are hardly any signals there.
So I am not sure that it is even possible to have the same kind of behavior.
(To react to a trigger.)

monsta, 2017-10-25:
I'll put it on my list of things to fix before 1.20 release.

info-cppsp, 2017-10-26:
https://www.freedesktop.org/software/gstreamer-sdk/data/docs/2012.5/gio/gdbus.html
[image]
If you look here, I have signed up for all emitted signals from UDisks2.
Cooling my HDD with a fan, I only see change in hddtemp, but no signal from UDisks2 is emitted.
UDisks2 only signals on the regular 10 minute refresh.

As stated before the only way I can see to solve this is to force a smart refresh, but IMHO that's not something we should do.

Anyway I'm looking forward to your solution. ;-)

monsta, 2017-10-26:
Well... when user sets refresh interval to 2 seconds in the prefs dialog, he expects each sensor to refresh every 2 seconds.

info-cppsp, 2017-10-26:
Well, yeah...
They are refreshed, just the underlying structure isn't...
But I know what you mean.

I have just checked on my laptop. mint 17.3 32 bit
Udisks works the same way.
Even if there is a 'dedicated' Changed signal, that also only fires every **30** !! minutes.
I have waited half an hour...
So at least, there is some improvement...

So you can choose this simple plugin, that works with SSDs, but updates temp every 10 minutes
OR
you can use hddtemp with a daemon, that doesn't work with SSDs, but updates on temp change...
OR
you can use this for SSDs, hddtemp for HDDs, until they fix SSD support

info-cppsp, 2018-01-30:
another note on the matter
https://github.com/mate-desktop/mate-sensors-applet/commit/cba9085981319f697440485f990e6b4961154c70#commitcomment-27208405

raveit65, 2018-02-12:
I really would like to see that Gui use the same settings as we have with the udisk2 plugin.