mate-desktop/mate-sensors-applet #73: Add sensor based on executable shell command

info-cppsp, 2018-01-26:
feature request (or idea) from raveit65

https://github.com/mate-desktop/mate-sensors-applet/issues/56#issuecomment-340683757

[test.c.txt](https://github.com/mate-desktop/mate-sensors-applet/files/1668339/test.c.txt)
```
gcc -Wall -g test.c -o test `pkg-config --cflags --libs glib-2.0 gobject-2.0`
```

A first attempt based on:
http://www.cs.uleth.ca/~holzmann/C/system/shell_commands.html
http://man7.org/linux/man-pages/man3/popen.3.html

Works, but very limited.

My problem is that hddtemp and smartctl need sudo to read the temperature...

raveit65, 2018-01-27:
I think we should leave that for next dev cycle.
I am complete busy with other releasing stuff.
Sadly, my old 3ware hardware-raid-controller-card is incompatible with my new mobo, so i can't test old scripts for reading temps of disks behind a hardware-raid array.
With the card i used commands like this with a script.
```
#!/bin/bash

smartctl -d 3ware,0 -A /dev/twa0 | grep Temperature_Celsius | awk '{print $10}'
```
I used the script with the command applet from xfce at this time.
I recal that i changed permissions  of /dev/twa0 device with the rc.local script to make it readable for my common user.
So this should be the way to test temps from smartctl or hdtemp with a command.

info-cppsp, 2018-01-27:
I don't mean this for 1.20 right now...
OK, If I don't have to worry about the permissions, this shouldn't be so difficult.

raveit65, 2018-02-27:
I am not really sure any more about this plugin.
As personal i don't use a hardware-raid-controller.
And i don't know if this is unsecure to suggest a user to change permission for devices /dev/<device>
@lukefromdc 
What do you think?

lukefromdc, 2018-02-27:
Not sure about this, but I suspect if a user could change ALL the permissions in /dev, then that user would gain access to all devices on the box and I suspect there would somehow be a way to leverage that into privilige escalation by gaining user priviliges on / as / is on a device listed in /dev . Thus, this needs more research.

info-cppsp, 2018-02-27:
How about instead of allowing a full shell command, one could only set up a file.
This would then be read by the plugin every refresh time for a double.
How the user gets a double value into the file is not my problem.

We could write an example in the documentation to set up a cron job to write to the file...

closed 2023-08-28 by raveit65