mate-desktop/mate-sensors-applet #76: V1.2

Ntemis, 2018-02-10:
#### Expected behaviour
To show hdd temps

#### Actual behaviour
even if i have a 
[image]
They aren't actually been used 

shows only cpu temp
#### Steps to reproduce the behaviour
you know

#### MATE general version
1.2

#### Package version

dpkg whatever
#### Linux Distribution
xenial 16.04.3

#### Link to downstream report of your Distribution
have no idea what this is

info-cppsp, 2018-02-10:
[image]

?

raveit65, 2018-02-10:
Sorry, we don't support Mate-1.2 any more,  since several years.
Actual version is mate-1.20.
So, file out a report against your distro and ask distro maintainer to fix that, or use a distro with actual Mate version.

raveit65, 2018-02-10:
What is so hard to post correct package versions for an ubuntu release?

Ntemis, 2018-02-11:
Ah sorry about this i was missing/failed to see there was an enable tick box on my 4k monitor lol
Thanks

closed 2018-02-11 by Ntemis