mate-desktop/mate-sensors-applet #81: Ati plugin is obsolete since a long time

raveit65, 2018-03-01:
#### MATE general version

1.20
#### Package version

1.20.1
#### Linux Distribution

fedora
#### Link to downstream report of your Distribution
I don't file out bugs against myself :D

Ati plugin is obsolete since a long time, the proprietary driver is given up by AMD since 2016.
Current open-source drivers are xorg-x11-drv-ati (7.9.0) or new AMD driver xorg-x11-drv-amdgpu.
I never build the ati plugin for rpmfusion as Catalyst or fglrx drivers were well known as buggy in fedora.
Those drivers are removed from rpmfusion since 2 years or so.

Maybe it's time to drop those plugin?
Btw. maybe it's possible to add support for new drivers.

Note:
Sorry , i am using always nvidia cards, so i can't help or test something.

monsta, 2018-03-01:
Right, I completely forgot about it. For 1.18, the last supported distro with [that driver](https://packages.debian.org/source/jessie/fglrx-driver) was LMDE 2, based on Debian Jessie. But we stopped building the plugin for it since 1.16 anyway (https://github.com/mate-desktop/debian-packages/commit/66255a73ea858f9e694fe299e46aceb92cf0bf11). Now 1.20 doesn't support LMDE 2 anyway, so I'm fine with dropping the plugin.

What about other distros?

monsta, 2018-03-01:
Oh wow, fglrx doesn't even support Xorg 1.18. Any supported distros left with 1.17...?

raveit65, 2018-03-01:
Rhel-7.3 use xorg-server-1.19.3.
I do not really think that there is one supported distro with xorg-server-1.17 when we release 1.22 in a year.
It's more a question if we have the manpower/hardware to support current xorg-x11-drv-amdgpu, available since ubuntu-16-04 or current fedora versions.

monsta, 2018-03-01:
It's available alright, but I have no idea how to get sensors info from it...

lukefromdc, 2018-03-01:
I have r600 ATI/AMD cards but use only open drivers. Catalyst/fglrx is not only closed but also has been caught and possibly passed in performance by the r600g driver according to numerious stories on Phoronix

info-cppsp, 2018-03-06:
https://www.gamingonlinux.com/wiki/Querying_and_benchmarking_AMD_GPUs
does this work for amdgpu?
`sudo cat /sys/kernel/debug/dri/0/amdgpu_pm_info`

radeon has sg similar, but no temperature info inside
sensors show a pci sensor for my old APU, but it is at -1.0 C, not sure what that means...

lukefromdc, 2018-03-06:
I have the older r600 driver and hardware and thus would not have that. Do we have someone testing on amdgpu?

monsta, 2018-03-16:
Interesting. We need someone with (new enough) AMD/ATI card to test this.

raveit65, 2023-08-28:
removed https://github.com/mate-desktop/mate-sensors-applet/commit/239212e11dcb3f7f8092417d7dd444e898a5c952

closed 2023-08-28 by raveit65

lukefromdc, 2023-08-29:
If FGLRX won't work with any xserver that would ever be used with MATE 1.28, then we lose absolutely nothing by removing this as anyone using it will be on a distro shipping an older version of MATE.