mate-desktop/mate-sensors-applet #84: NVIDIA Fan Speed showing percentage speed instead of RPM

stephenangelico, 2018-06-28:
#### Expected behaviour
Values between 0 and 3000ish

#### Actual behaviour
Values between 0 and 40

#### Steps to reproduce the behaviour
- Add MATE Sensors Applet to a panel
- Enable CoolerLevel for an NVIDIA GPU
- Observe reported fan speed as GPU fan spins up

#### MATE general version
1.18.0

#### Package version
1.18.2-5ubuntu0

#### Linux Distribution
Ubuntu MATE 17.10

I discovered that `nvidia-settings` has, among its query options, `GPUCurrentFanSpeed`, which returns a percentage, and `GPUCurrentFanSpeedRPM`, which returns a believable number or RPM. The command `nvidia-settings -tq :0/GPUCurrentFanSpeedRPM` returns a single value that should be the RPM of the GPU.

raveit65, 2018-06-28:
I can't test this as my nvidia card is fanless.

danielw2904, 2018-06-30:
The plugin takes the cooler level as input it seems (see below). Can someone tell me where to find the appropriate nvidia header files?
I'll test this when I get to my PC. 
```if (XNVCTRLQueryTargetCount(nvidia_sensors_dpy,
                        NV_CTRL_TARGET_TYPE_COOLER,
                        &cnt)) {

            for (i = 0; i < cnt; i++) {
                gchar *id = g_strdup_printf("GPU%d%s", i, THERMAL_COOLER_LEVEL);

                sensors_applet_plugin_add_sensor(&sensors,
                                                 THERMAL_COOLER_LEVEL,
                                                 id,
                                                 _("GPU"),
                                                 FAN_SENSOR,
                                                 TRUE,
                                                 FAN_ICON,
                                                 DEFAULT_GRAPH_COLOR);

                g_free(id);
            }
        }
```

monsta, 2018-08-19:
My GT 730 has a fan, but I can't find it in the sensors list anymore. I remember the fan was there before, now I have only two temp sensors which show the same value.

I don't think we made any related changes here, so it must be something in nvidia libs.

Tested in Mint 19 (based on Ubuntu 18.04).

monsta, 2018-08-20:
Looks like there were some changes in the past years...

https://github.com/NVIDIA/nvidia-settings/commit/db3b5f655464386838d0c5859cddaf6c4f812979#diff-62dcb5f0ffc3089b7565897d8beb3322L177
https://github.com/NVIDIA/nvidia-settings/commit/1f505d9ccb4106a8bcf0d539ac432c716ebc7336#diff-62dcb5f0ffc3089b7565897d8beb3322L182

stephenangelico, 2018-08-20:
So, big surprise, NVIDIA broke it. Does this mean to properly close this, we'll need to check which version of the driver is installed in case of legacy devices?

raveit65, 2018-08-20:
Why nvidia broke it?
I assume that that we need to build against a higher version for nvctrl.

monsta, 2018-08-21:
Ok, I thought it was working until some point. Looks like it's not so.
Normally I don't have any fan sensors enabled in the applet, so no wonder I forgot about my own findings.

It was the same from the start:
https://github.com/mate-desktop/mate-sensors-applet/pull/30#issuecomment-118330280
https://github.com/mate-desktop/mate-sensors-applet/pull/30#issuecomment-118829087

Sorry about my memory :slightly_smiling_face:

monsta, 2018-08-21:
Actually, https://github.com/mate-desktop/mate-sensors-applet/pull/30#issuecomment-118330280 is still reproducible. It's another GT 730 which I have on my work machine. It only exposes fan speed in percents, as can be seen on the screenshot.

monsta, 2018-08-27:
@stephenangelico @danielw2904 please check out https://github.com/mate-desktop/mate-sensors-applet/pull/86 for possible fix.

stephenangelico, 2018-08-28:
I don't have a build environment configured so it's not easy for me to test. If I could simply get a replacement `libnvidia.so` (or whichever file that commit changes) then I'd be happy to test that.

monsta, 2018-08-28:
Do you still use Ubuntu 17.10? Because if you have already upgraded to 18.04, I can upload a patched mate-sensors-applet package to my PPA for testing. (17.10 is [EOL](https://lists.ubuntu.com/archives/ubuntu-security-announce/2018-July/004483.html) now anyway.)

stephenangelico, 2018-08-28:
No, since opening this issue I have upgraded to 18.04.

stephenangelico, 2018-09-14:
Any further news on that patch? I saw changes requested on #86 and I haven't seen any patched package (nor have I been sure I found @monsta 's PPA).

monsta, 2018-09-17:
Yeah, sorry, I should've packed it for testing... I'll do it tomorrow.

Changes are requested for the case when the fan is present but getting speed in RPMs is unsupported. When it's supported, the patch should work fine.

stephenangelico, 2018-10-06:
Sorry for taking so long with this, I had a lot on and also pushed back testing "until the next time I log on" which for me is rare. Anyways...
[image]
It works! I haven't done much testing, I simply fired up an inefficient game that could quickly warm up the GPU and compared the numbers I was seeing to those in `nvidia-settings`. Looks good to me!

I won't close yet in case there are problems merging this patch into master, and in case there are other things I should test. Please note that I do not want to have to downgrade my driver (if older drivers are problematic for some people), and in fact I can't go back further than NVIDIA driver 375 as that appears to be the first driver with GTX 1060 support.

monsta, 2018-10-06:
Ok, good. I need to get back to the PR to address the issues...