mate-desktop/mate-sensors-applet #97: There is no CPU sensor on Mate Sensors Applet

LightColoredLines, 2019-09-11:
#### Expected behaviour
libsensors should be in there to show CPU temp.

#### Actual behaviour
There is no libsensors to show CPU temp. Only Udisks2 and Nvidia.

#### Steps to reproduce the behaviour
Add Hardware Sensors Monitor to a panel

#### MATE general version
MATE 1.20.4

#### Package version
1.20.2

#### Linux Distribution
Antergos 64 bit 4.20.12-arch1-1-ARCH mate-desktop/mate-applets#1 SMP PREEMPT

#### Screenshot
[image]

raveit65, 2023-08-28:
Does the issue still exists?

bjo81, 2024-11-07:
I have a similar issue on Arch - only the GPU is shown. But as libsensors is used, I don't think it's an applet rebuild issue.

[image]

But sensors show also nvme and k10:

```
nvme-pci-0100
Adapter: PCI adapter
Composite:    +39.9°C  (low  = -273.1°C, high = +80.8°C)
                       (crit = +84.8°C)
Sensor 1:     +39.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +45.9°C  (low  = -273.1°C, high = +65261.8°C)

amdgpu-pci-0a00
Adapter: PCI adapter
vddgfx:        6.00 mV 
fan1:           0 RPM  (min =    0 RPM, max = 3300 RPM)
edge:         +34.0°C  (crit = +110.0°C, hyst = -273.1°C)
                       (emerg = +115.0°C)
junction:     +36.0°C  (crit = +110.0°C, hyst = -273.1°C)
                       (emerg = +115.0°C)
mem:          +32.0°C  (crit = +105.0°C, hyst = -273.1°C)
                       (emerg = +110.0°C)
PPT:           4.00 W  (cap = 211.00 W)

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +42.0°C 
```