mate-desktop/mate-sensors-applet #98: Checking disk in udisks section results in wrong vertical arrangement of sensor icons

N0rbert, 2019-09-11:
#### Expected behaviour
all sensors meters are shown in horizontal line

#### Actual behaviour
only one sensor is shown, other are "below" it (see attached screencast)

[image]

#### Steps to reproduce the behaviour
1. Install Ubuntu MATE 19.10
2. Login to installed system
3. Add Sensors Applet to the panel
4. Open Sensors Applet Preferences, go to Sensors tab
5. Check *Disk Temperature* under udisks2 section

#### MATE general version
1.22.1

#### Package version
1.22.1-1

#### Linux Distribution
Ubuntu-MATE 19.10 "Eoan Ermine" - Alpha amd64 (20190909)

#### Link to downstream report of your Distribution
http://bugs.launchpad.net/bugs/1843395

lukefromdc, 2019-09-11:
I've sometimes had the same problem restarting the panel with` killall mate-panel` (it will then automatically restart) will cause them to show up side by side. CONFIRMED.

N0rbert, 2021-10-29:
Still happens in MATE 1.26.0.

maravento, 2022-12-22:
In ubuntu mate 1.26.0 the udisk sensor doesn't even appear