mate-desktop/mate-sensors-applet #104: When MATE is set to Turkish, libsensors returns an error

cardpuncher, 2020-05-02:
When MATE is set to use Turkish, libsensors returns an error and can't read the temperature. mate-sensors-applet simply displays "HATA" which means "Error" in Turkish. In English or French, everything works fine. Please see the attached screenshot for an example.

#### MATE general version
1.24.0

#### Package version
1.24.0-1

#### Linux Distribution
Ubuntu

#### Link to downstream report of your Distribution
https://bugs.launchpad.net/ubuntu/+source/mate-sensors-applet/+bug/1794348

[image]

raveit65, 2020-06-05:
Hi,
i checked build log of 1.24 release but i didn#t see a build warning for translations of your language.
https://travis-ci.org/github/mate-desktop/mate-sensors-applet/jobs/649558212
I am using sensor-applet too with amd and nvidia hardware.
So, i can switch tomorrow morning (when i am fresh :) ) my desktop to Turkish language, to see if i can reproduce the issue.
Not sure if i can debug something because i am in doubt that i understand my desktop in Turkish :)

But i recall such a temp1 warning sometimes. It seems that temp1 is my wifi card. So every time i disable wifi with network-manager i get a temp1 warning.

raveit65, 2020-06-06:
Ok, it happens with all sensors from libsensors and udisk2 plugin. Only the nvidia plug-in is working here.
[image]
With that error message
```
rave@satellite ~]$ LANG=tr_TR
[rave@satellite ~]$ /usr/libexec/mate-sensors-applet

(mate-sensors-applet:4025): Gtk-WARNING **: 11:03:28.221: Invalid input string

(mate-sensors-applet:4025): Gtk-WARNING **: 11:03:28.221: Invalid input string
```
I guess it was a problem with migration from intltools to gettext.
@mate-desktop/core-team 
Can you please take a look into it. Issue is posted from our transifex admin (team).

raveit65, 2020-06-08:
The tooltip of the sensors shows me `Errror compiling URL regex`

raveit65, 2020-06-08:
libsensors plugin looks at this place for the `url`.
https://github.com/mate-desktop/mate-sensors-applet/blob/master/plugins/libsensors/libsensors-plugin.c#L199
`url = g_strdup_printf ("sensor://%s/%d", chip_name, data->number);`
and the tooltip error message is here.
libsensors
`g_set_error (error, SENSORS_APPLET_PLUGIN_ERROR, LIBSENSORS_REGEX_URL_COMPILE_ERROR, "Error compiling URL regex");`

I guess something is wrong with reading the vars from url.

raveit65, 2020-06-08:
@cardpuncher 
Do you have this sympol errors when calling lm_sensors directly in terminal?
```
[rave@satellite ~]$ sensors
BAT1-acpi-0
Adapter: ACPI interface
in0:          12.53 V  
curr1:         0.00 A  

coretemp-isa-0000
Adapter: ISA adapter
Core 0:       +39.0�C  (high = +95.0�C, crit = +105.0�C)
Core 2:       +35.0�C  (high = +95.0�C, crit = +105.0�C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +43.0�C  (crit = +100.0�C)
```

raveit65, 2020-06-08:
Hmm, with fr_FR language i get this warning too
`(mate-sensors-applet:4025): Gtk-WARNING **: 11:03:28.221: Invalid input string`
but all sensors are displaying fine.

I checked tr.po but i didn't saw any error.

rbuj, 2020-06-08:
There is an error in the translation, but I don't think it's related to the runtime error, tomorrow I will debug the code in order to find the source of the error as I can't do it today.
```
$ msgfmt -c --check-accelerators='_' po/tr.po 
po/tr.po:409: msgstr lacks the keyboard accelerator mark '_'
msgfmt: found 1 fatal error
```
Missing accelerator '_' at po/tr.po:409:
```
#: sensors-applet/sensor-config-dialog.c:651
msgid "Alarm _repeat interval (secs)"
msgstr "Alarmın tekrarlanma aralığı (saniye)"
```
@raveit65 I don't think the error is located on [libsensors-plugin.c:199](https://github.com/mate-desktop/mate-sensors-applet/blob/master/plugins/libsensors/libsensors-plugin.c#L199), because SENSORS_API_VERSION > 0x400
```
$ grep SENSORS_API_VERSION /usr/include/sensors/sensors.h
#define SENSORS_API_VERSION		0x500
```

cardpuncher, 2020-06-08:
8.06.2020 11:47 tarihinde raveit65 yazdı:
> @cardpuncher <https://github.com/cardpuncher>
> Do you have this sympol errors when calling lm_sensors directly in terminal?
>
> |[rave@satellite ~]$ sensors BAT1-acpi-0 Adapter: ACPI interface in0: [...]

Everything seems to be OK when calling lm_sensors directly from the
terminal, here's the output:

$ sensors
fam15h_power-pci-00c4
Adapter: PCI adapter
power1:        7.90 W  (interval =   0.01 s, crit =  14.98 W)

radeon-pci-0100
Adapter: PCI adapter
temp1:            N/A  (crit = +120.0°C, hyst = +90.0°C)

BAT1-acpi-0
Adapter: ACPI interface
in0:          16.55 V
curr1:       1000.00 uA

k10temp-pci-00c3
Adapter: PCI adapter
temp1:        +51.5°C  (high = +70.0°C)
                       (crit = +115.5°C, hyst = +108.0°C)

radeon-pci-0008
Adapter: PCI adapter
temp1:        +53.0°C  (crit = +120.0°C, hyst = +90.0°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +51.0°C  (crit = +226.8°C)
temp2:        +20.0°C

I added the keyboard accelerator on Transifex, sorry for forgetting it
when translating. Also, FYI, this bug was happening with Ubuntu 18.04 as
well, and it has MATE 1.20 if I remember correctly.

rbuj, 2020-06-09:
glibc's `regexec` returns 1 for Turkish locale, i.e. no match.
https://github.com/mate-desktop/mate-sensors-applet/blob/e4725256234330e5e649e772e5ff8f13b61aa438/plugins/libsensors/libsensors-plugin.c#L402

### Displaying the error message

[image]

```patch
--- a/plugins/libsensors/libsensors-plugin.c
+++ b/plugins/libsensors/libsensors-plugin.c
@@ -396,10 +396,11 @@ static gdouble libsensors_plugin_get_sensor_value(const gchar *path,
                                                     SensorType type,
                                                     GError **error) {
     gdouble result = 0;
+    int status;
     regmatch_t m[3];
 
     /* parse the uri into a (chip, feature) tuplet */
-    if (regexec (&uri_re, path, 3, m, 0) == 0) {
+    if ((status = regexec (&uri_re, path, 3, m, 0)) == 0) {
         const sensors_chip_name *found_chip;
         int feature;
 
@@ -426,7 +427,9 @@ static gdouble libsensors_plugin_get_sensor_value(const gchar *path,
             g_set_error (error, SENSORS_APPLET_PLUGIN_ERROR, LIBSENSORS_CHIP_NOT_FOUND_ERROR, "Chip not found");
         }
     } else {
-        g_set_error (error, SENSORS_APPLET_PLUGIN_ERROR, LIBSENSORS_REGEX_URL_COMPILE_ERROR, "Error compiling URL regex");
+        char buffer[100];
+        regerror(status, &uri_re, buffer, 100);
+        g_set_error (error, SENSORS_APPLET_PLUGIN_ERROR, LIBSENSORS_REGEX_URL_COMPILE_ERROR, "Error compiling URL regex: %s", buffer);
     }
     return result;
 }
```

[image]

closed 2020-06-11 by raveit65

cardpuncher, 2020-06-11:
On 10.06.2020 19:52, raveit65 wrote:
> @cardpuncher <https://github.com/cardpuncher>
> Do you like to test #107
> <https://github.com/mate-desktop/mate-sensors-applet/pull/107> ?
> It fixes for me the issue.

That nice to hear, thanks for the quick fix. As for testing #107, I'm
afraid it involves compiling MATE's source code and that this is well
beyond my skills, I'm just a translator.

But, if it's possible for you to create a .deb package, I would be happy
to test. If not, I would give a try as soon as a Live DVD of a distro
containing the fix is released.

PR #107 (2020-06-11): libsensors-plugin: Port to GRegex
Closes #104