mate-desktop/mate-sensors-applet #117: Graphs with the same unit should use the same scale

riccardomurri, 2020-12-09:
#### Expected behaviour

I would like to be able to compare readings on graphs showing temperature at a glance: if the bar on the disk temp is higher than the CPU temperature graph, then it should mean that the disk temperature is higher than the CPU's.

#### Actual behaviour

[Each graph is scaled separately according to the sensor's min and max values][1], making graphs useless without hovering for reading the actual values.  The screenshot shows a disk temperature graph (left) seemingly higher than the CPU temperature graph (right), but hovering reveals that the opposite is true: disk is at 45 °C and CPU is at 58 °C.

[image]

[1]: https://github.com/mate-desktop/mate-sensors-applet/blob/master/sensors-applet/active-sensor.c#L292

#### Steps to reproduce the behaviour

On my Thinkpad X1 Carbon, enable all temperature sensors from libsensors.

#### MATE general version

1.24.1

#### Package version

```
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                       Version      Architecture Description
+++-==========================-============-============-==========================================================================
ii  mate-sensors-applet        1.24.1-1     amd64        Display readings from hardware sensors in your MATE panel
ii  mate-sensors-applet-common 1.24.1-1     all          Display readings from hardware sensors in your MATE panel (common files)
```

#### Linux Distribution

Debian testing

#### Link to bugreport of your Distribution (requirement)

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=976961

raveit65, 2020-12-12:
I can confirm that something is wrong with the graph. One of my disks which has 30°C usually is shown higher than CPU temp 38°C, when i choose the graph, And my M2 disk won't shown.
All works fine  when i select icon + value, which is my preferred setting.
Btw. thew command `sensors` will show you the correct values in terminal.

rbuj, 2020-12-12:
It's not a bug, since all semiconductors have a specified [safe operating area](https://en.wikipedia.org/wiki/Safe_operating_area).
In screenshot, you only have to worry about the sensors which are filling the entire area of the graph.