mate-desktop/mate-sensors-applet #123: mate-sensors-applet-1.24.1 fails to build on systems with musl C library

ernsteiswuerfel, 2021-03-21:
I tried to build MATE desktop environment on Gentoo Linux with musl as system C library instead of glibc. At the moment it fails at building `mate-extra/mate-sensors-applet-1.24.1`. Tried mate-sensors-applet from `git master`, which shows the same failure:

#### Actual behaviour
```
[...]
make[2]: Leaving directory '/root/build/mate-sensors-applet/plugins/udisks2'
Making all in i2c-proc
make[2]: Entering directory '/root/build/mate-sensors-applet/plugins/i2c-proc'
  CC       i2c-proc-plugin.lo
i2c-proc-plugin.c: In function 'i2c_proc_plugin_get_sensor_value':
i2c-proc-plugin.c:147:31: warning: implicit declaration of function 'setlocale' [-Wimplicit-function-declaration]
  147 |     if (NULL == (old_locale = setlocale(LC_NUMERIC, "C"))) {
      |                               ^~~~~~~~~
i2c-proc-plugin.c:147:41: error: 'LC_NUMERIC' undeclared (first use in this function)
  147 |     if (NULL == (old_locale = setlocale(LC_NUMERIC, "C"))) {
      |                                         ^~~~~~~~~~
i2c-proc-plugin.c:147:41: note: each undeclared identifier is reported only once for each function it appears in
i2c-proc-plugin.c: At top level:
i2c-proc-plugin.c:214:14: warning: no previous prototype for 'sensors_applet_plugin_name' [-Wmissing-prototypes]
  214 | const gchar *sensors_applet_plugin_name(void)
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [Makefile:497: i2c-proc-plugin.lo] Error 1
make[2]: Leaving directory '/root/build/mate-sensors-applet/plugins/i2c-proc'
make[1]: *** [Makefile:426: all-recursive] Error 1
make[1]: Leaving directory '/root/build/mate-sensors-applet/plugins'
make: *** [Makefile:574: all-recursive] Error 1
```

#### Steps to reproduce the behaviour
git clone https://github.com/mate-desktop/mate-sensors-applet/
cd mate-sensors-applet/
./autogen.sh
make

#### MATE general version
1.24

#### Package version
1.24.1, git-master

#### Linux Distribution
Gentoo Base System release 2.7
musl-1.2.2

#### Link to bugreport of your Distribution (requirement)
https://bugs.gentoo.org/777375
[build.log](https://github.com/mate-desktop/mate-sensors-applet/files/6177992/build.log)

ernsteiswuerfel, 2021-03-21:
Seems this is about the part
```
#ifdef HAVE_LOCALE_H
#include <locale.h>
#endif

```
in **plugins/i2c-proc/i2c-proc-plugin.c**. 

If I delete the ifdef and the endif line, to unconditionally inlude <locale.h> the build succeeds.

rbuj, 2021-03-22:
Please, can you send a pull request?

```
# There is no longer any need to check for headers that are part of
# ISO C90 (as amended): assert.h, ctype.h, errno.h, float.h, iso646.h,
# limits.h, locale.h, math.h, setjmp.h, signal.h, stdarg.h, stddef.h,
# stdio.h, stdlib.h, string.h, time.h, wchar.h, wctype.h.
```
https://git.savannah.gnu.org/cgit/autoconf.git/tree/lib/autoconf/headers.m4#n358

raveit65, 2023-08-28:
https://github.com/mate-desktop/mate-sensors-applet/commit/fc56df38106f664def119611d302d20d0ea65e67

closed 2023-08-28 by raveit65