mate-desktop/mate-sensors-applet #129: Doesn't show udisk temperature sensor

maravento, 2022-12-22:
#### Expected behaviour

show udisk temperature sensor

#### Actual behaviour

Doesn't show **udisk** temperature sensor

[image]

#### Steps to reproduce the behaviour

#### MATE general version

1.26.0

#### Linux Distribution

Ubuntu Mate 22.04.1

N0rbert, 2023-08-05:
Confirmed on Debian 11 with mate-sensors-applet 1.24.1-1. It does not show the temperature of my NVME SSD.
My device has only one NVME disk, so I proceed with the following.
I found the following workaround - I have added *Command* applet to MATE Panel and placed the following command

```
bash -c "inxi -xD -c0 | grep 'temp.*' --only-matching | tail -n1 | sed 's/temp:/🖴/' | sed 's/C/°C/'"
```

into *Command* field.

lukefromdc, 2023-08-05:
If the problem is in libatasmart there's nothing we can do about it here. That said, the issue report there says it could fix this and similar problems in several other packages

maravento, 2023-08-05:
> If the problem is in libatasmart there's nothing we can do about it here. That said, the issue report at [Rupan/libatasmart#9](https://github.com/Rupan/libatasmart/pull/9) says it could fix this and similar problems in several other packages

I don't think so. this problem has been unresolved for years https://bugs.launchpad.net/ubuntu/+source/libatasmart/+bug/1581594

lukefromdc, 2023-08-05:
If they won't fix it, we cannot fix it for them unless someone want to reimplement their work or find a way to bypass it. I don't know enough about reading the sensors to implement such a workaround myself. Looks like libatasmart in turn is choking on a problem in udisks2. 

Anyone able to code a fix should submit it as a PR to libatasmart instead of waiting for them to fix it. As a DE with a small team, we don't have the peoplepower to fork things like libatasmart much less udisks2 over this.

I get these errors in the output of running `sensors` , no idea what the actual temperature of that device sitting at idle is. Wonder if libatasmart is erroring out on not being able to see that minimum temperature rather than ignoring it and moving on?

nvme-pci-0100
Adapter: PCI adapter
ERROR: Can't get value of subfeature temp1_min: I/O error
Composite:    +47.9°C  (low  =  +0.0°C, high = +99.8°C)
                       (crit = +119.8°C)
ERROR: Can't get value of subfeature temp2_min: I/O error
Sensor 1:     +47.9°C  (low  =  +0.0°C, high = +65261.8°C)

maravento, 2023-08-05:
> libatasmart

check this alternative https://github.com/storaged-project/udisks/issues/433

N0rbert, 2023-08-05:
If we talk about `inxi -Dx` - it uses `nvme smart-log /dev/nvme0n1` under the hood to get disk info (including its temperature).  Another way to get NVME temperature is `smartctl -a /dev/nvme0n1`. 
So actually *mate-sensors-applet* may need new plugin for NVME.

I have also reported https://github.com/storaged-project/udisks/issues/1163 to udisks2 about temperature of NVME drives. Udisks2 does not read temperature of NVME drive, so it seems that *udisks2-plugin.c* of *mate-sensors-applet* works correctly (there is no data = nothing to show).

N0rbert, 2023-08-06:
I have digged a bit deeper and now I see the following on my Debian 11: 
NVME temperature(s) may be  obtained from *libsensors* plugin.
For example the below output from `lm-sensors`:
<pre>
$ sensors
iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +30.0°C  

nvme-pci-0100
Adapter: PCI adapter
<b>Composite</b>:    +28.9°C  (low  = -273.1°C, high = +81.8°C)
                       (crit = +84.8°C)
<b>Sensor 1</b>:     +28.9°C  (low  = -273.1°C, high = +65261.8°C)
<b>Sensor 2</b>:     +29.9°C  (low  = -273.1°C, high = +65261.8°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +35.0°C  (crit = +100.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +32.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +30.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +32.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +30.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +31.0°C  (high = +100.0°C, crit = +100.0°C)

BAT1-acpi-0
Adapter: ACPI interface
in0:          11.02 V  
curr1:       799.00 mA 
</pre>

means that user can view NVME temperature as *Composite* without any additional coding and so on:

[image]

I see this behavior on one modern 11th intel laptop and two desktops (11th intel and AMD Ryzen 5700G).

maravento, 2023-08-06:
almost hit

[image]

raveit65, 2023-08-06:
No idea what composite means. I think sensor 1 or 2 are valid and i use sensor 2 for displaying the temp in sensor-applet.
Can we close the issue?

raveit65, 2023-08-06:
Btw. sata disk temps are displayed with udisk2 in applet since more than many years.

tbzatek, 2023-08-07:
> No idea what composite means. I think sensor 1 or 2 are valid and i use sensor 2 for displaying the temp in sensor-applet. Can we close the issue?

Composite means kind of average of multiple sensors on the drive, if any. UDisks provides this value as the primary `org.freedesktop.UDisks2.NVMe.Controller.SmartTemperature` property: http://storaged.org/doc/udisks2-api/latest/gdbus-org.freedesktop.UDisks2.NVMe.Controller.html#gdbus-property-org-freedesktop-UDisks2-NVMe-Controller.SmartTemperature

maravento, 2023-08-07:
> Btw. sata disk temps are displayed with udisk2 in applet since more than many years.

And that is precisely why this issue is still open. Because the value provided by udisk2 does not correspond to reality and is always 98% or 100% with sata ssd.

tbzatek, 2023-08-07:
> > Btw. sata disk temps are displayed with udisk2 in applet since more than many years.
> 
> And that is precisely why this issue is still open. Because the value provided by udisk2 does not correspond to reality and is always 98% or 100% with sata ssd.

That's https://github.com/storaged-project/udisks/issues/568 on a short-term TODO list.

maravento, 2023-08-07:
> 

another bug reported in 2018 and we are in 2023 with no solution...

PS: If I knew what the solution is, I would have already pushed a pull request, but unfortunately I don't know how to fix it. That's why I posted a workaround bash script https://github.com/mate-desktop/mate-sensors-applet/issues/129#issuecomment-1666912045

maravento, 2023-08-07:
At least, as N0rbert says in https://github.com/mate-desktop/mate-sensors-applet/issues/129#issuecomment-1666867394. This is a great advance.

raveit65, 2023-08-07:
good, closing.

closed 2023-08-07 by raveit65

antoniovazquezblanco, 2024-05-05:
This has been fixed in an independent fork of the original lib that it is not currently recognized as official upstream by maintainers at https://github.com/libatasmart/libatasmart.