mate-desktop/mate-sensors-applet #134: Mellanox Virtual Card Temperature Value not displayed ERROR

tonyabc, 2023-10-15:
#### Expected behaviour
The Mellanox Range of Network Adapters have a temperature sensor that outputs it's value in Centigrade and displays using the Sensors command successfully.  The sensor is detected in Mate-Sensors-Applet as "Temp1" variable and selecting it should generate a corresponding value on the panel.
Sensors output
`
mlx5_0000:a2:00.0-virtual-0
Adapter: Virtual device
temp1:        +41.0°C  
`

#### Actual behaviour
When the "Temp1" variable is selected in the mate-sensors-applet preferences selection list, instead of producing a value it instead produces an ERROR flag and in notifications "Error updating sensor Temp1 - An error occurred while trying to update......" notification.  Hovering over the ERROR flag in the panel shows "Temp1 - error compiling URL regex : Not Match"

#### Steps to reproduce the behaviour
Install Mellanox card, select Temp1 variable relating to value in mate-sensors-applet preferences.

#### MATE general version
Version 1.26.0

#### Package version
1.26

#### Linux Distribution
Fedora Compiz/Mate spin.

#### Link to bugreport of your Distribution (requirement)
https://bugzilla.redhat.com/show_bug.cgi?id=2244258

lukefromdc, 2023-10-16:
There may be more than one TEMP1 in mate-sensors applet with your install, and it's been so long since the way the applet interacts with libsensors has been updated that this list is often totally different that what running `sensors` shows. I have one of several "TEMP3" values as my CPU temp, it matches` k10temp-pci-00c3` and either `Tctl `or` Tccd1` nmot sure which as they are so close. It's a LOT of trial and error with varying CPU and GPU loads to get this right on setting the applet up anew.

Look through every temp value the applet offers for one matching the output from
`mlx5_0000:a2:00.0-virtual-0 Adapter:` it's a good chance that's the right one.

tonyabc, 2023-10-25:
Hi

It's the correct sensor as the notification popup actually states that it's the mlx5 adapter that's erroring.