mate-desktop/mate-sensors-applet #135: udisks2 disappeared after upgrading to 6.10.3 kernel

pkzc, 2024-08-10:
#### Expected behaviour
With Linux kernel 6.9.12 and eariler mate-sensors-applet 1.28.0 had two sensor groups shown in the Preferences panel: libsensors and usisks2.

#### Actual behaviour
After upgrading to kernel 6.10.3 only libsensors is shown, no udisks2. When I boot 6.9.12, udisks2 comes back.

#### Steps to reproduce the behaviour

#### MATE general version
mate-desktop   1.28.2-1.fc40.x86_64

#### Package version
mate-sensors-applet   1.28.0-1.fc40.x86_64

#### Linux Distribution
Fedora Linux 40

#### Link to bugreport of your Distribution (requirement)
https://bugzilla.redhat.com/show_bug.cgi?id=2303937

lukefromdc, 2024-08-10:
This will be a kernel isse and should not relate directly to MATE. Check to see if udisks sensors can be found from the command line via the "sensors" program. We do need to know whether the interface changed or support is broken.

pkzc, 2024-08-21:
With kernel 6.10.5-200.fc40.x86_64 x86_64 the problem is gone, udisks2 is back on the sensor list.

lukefromdc, 2024-08-21:
Confirms this was kernel and unrelated to MATE's frontend for this

closed 2024-08-21 by lukefromdc