mate-desktop/mate-sensors-applet #136: Need Help: Stuck during upgrade from V1.24.0-1 to V1.28.0

OS2User, 2024-12-15:
I didn't find any other place to ask for help, so I am doing it here.

Only available mate-sensors-applet in the repository is V1.26.0,-1 which is acting very weird when it comes to changing the colour of the graphs for most of my temperature sensors. And I just can't take in any longer!

It is definitively NOT expected behaviour, that the graph turns red already, when the actual temperature, for example of the Intel PCH, drops to < 25 K of the set "Sensor high limit" value of 88°C and I also don't consider it normal, that the graph is already orange at 25 K below the set "Sensor high limit". 

25 K below the set limit is normal operating range and should not warrant any orange or red graph whatsoever. 
Turning orange maybe some 7-5 K below and turning red up to 2 K below the limit would seem to be much more appropriate and go along with daily life common sense.

Anyhow, following the link in the V1.26.0-1 Help to the project's homepage led me here, where I found the latest version to be V1.28.0 [https://github.com/mate-desktop/mate-sensors-applet/releases/tag/v1.28.0](url), which I downloaded and tried to install in the hopes, that this weird graph colorization issue is fixed, after not having seen anything obvious here in the README, suggesting not to do so.

#### Expected behaviour
That the upgrade completes and replaces the existing V1.26.0-1 `/usr/lib/mate-sensors-applet/mate-sensors-applet`
And that after a successful update the color changes of those graphs behave more sane and in the range of common expectation.

#### Actual behaviour
`$ ./autogen.sh --prefix=/usr` 	→ 	`You need to install mate-common`  → V1.26.0-1 via Synaptic 
`$ ./autogen.sh --prefix=/usr	/usr/bin/mate-autogen	You need to install yelp-tools`  → V42.0-1 via Synaptic 
`$ ./autogen.sh --prefix=/usr
/usr/bin/mate-autogen
/usr/bin/yelp-build
checking for autoreconf >= 2.53...
  testing autoreconf... found 2.71
configure.ac:16: warning: The macro `AC_HEADER_STDC' is obsolete.
configure.ac:16: You should run autoupdate.
./lib/autoconf/headers.m4:704: AC_HEADER_STDC is expanded from...
configure.ac:16: the top level
configure.ac:19: warning: The macro `AC_PROG_LIBTOOL' is obsolete.
configure.ac:19: You should run autoupdate.
m4/libtool.m4:100: AC_PROG_LIBTOOL is expanded from...
configure.ac:19: the top level
checking for pkg-config >= 0.14.0...
  testing pkg-config... found 0.29.2
checking for mate-common >= 1.1.0...
  testing mate-doc-common... found 1.26.0
Checking for required M4 macros...
Checking for forbidden M4 macros...
Processing ./configure.ac
configure.ac:16: warning: The macro `AC_HEADER_STDC' is obsolete.
configure.ac:16: You should run autoupdate.
./lib/autoconf/headers.m4:704: AC_HEADER_STDC is expanded from...
configure.ac:16: the top level
configure.ac:19: warning: The macro `AC_PROG_LIBTOOL' is obsolete.
configure.ac:19: You should run autoupdate.
m4/libtool.m4:100: AC_PROG_LIBTOOL is expanded from...
configure.ac:19: the top level
Running mate-doc-common...
Running autoreconf...
autoreconf: export WARNINGS=no-portability
autoreconf: Entering directory '.'
autoreconf: running: autopoint --force
Can't exec "autopoint": No such file or directory at /usr/share/autoconf/Autom4te/FileUtils.pm line 293.
autoreconf: error: autopoint failed with exit status: 2`

So, what does this all mean so far and where or what is this "You should run autoupdate."? Autoupdate of what?
I certainly didn't find it anywhere in the unpacked `/mate-sensors-applet-1.28.0`.

#### Steps to reproduce the behaviour
Rerun the update

#### MATE general version
MATE 1.26.0 on x11

#### Package version
mate-sensors-applet 1.28.0 release

#### Linux Distribution
Linux Mint 21.3 Mate (based on Ubuntu 22.04 Jammy)

#### Link to bugreport of your Distribution (requirement)
[https://github.com/mate-desktop/mate-sensors-applet/issues](url)

lukefromdc, 2024-12-15:
You need to install mate-common first, the version provided through synaptic should
work but if it does not, build and install it from source first.That's why you had a file
not found.

The "autoupdate" stuff has nothing to do with it. I see that and always ignore it
as I simply don't know enough about Autotools to work on that.

lukefromdc, 2024-12-23:
I actually have only ever installed mate-common from the unpacked tarball
(however downloaded) by running
`autogen -sh --prefix=/usr` (since I install to the main /usr directories and even pack the results into .debs)
thus i cannot vouch for anything else 
`install-sh` is generated by running `autogen -sh` and presumably is then used by `make install` which is run after `make`

So to install for use with your existing MATE install you would run
`autogen -sh --prefix=/usr` 
then
`make`
then 
`make install`
You should then be able to build other MATE packages. In most cases you will need the -dev packages for some of the other MATE packages, gtk3, and glib as well. In this case you will need libmate-panel-applet-dev plus the gtk, glib, and libnotify development (-dev) packages

lukefromdc, 2024-12-23:
Also note that the last change to tooltip-handing code in this applet was in 2016 and in mate-panel was in 2018. While I do have the tooltips working here, I am using the wayland session and almost certainly a different GTK3 version. Tooltip handling is for the most part done inside GTK,

OS2User, 2024-12-23:
Oh dear! 
This upgrade attempt from mate-sensors-applet 1.26.0 to 1.28.0 seems to be turning into a major undertaking and as a senior citizen user only, I'm way out of my depth here.
So I think the smartest thing to do is to just wait a little longer until I can upgrade to LM 22.0 and hopefully find V1.28 deployed and most/all of my stated issues being gone there with it.

Btw, I fixed the link to mate-common-1.28.0.tar.xz in my previous post
[https://github.com/mate-desktop/mate-sensors-applet/issues/136#issuecomment-2558696105](url). 
I certainly didn't mean to express that I am trying to run the scripts from the webpage rather than from within the downloaded package ;-)

From all the items I have added to my bottom Panel, mate-sensors-applet so far is the only one failing to handle the tooltips properly.